bennymeg/Fabrication-Toolkit #2: No Dialog on M1 Mac

wraybowling, 2022-07-08:
As far as I can tell, to use this plugin I'm supposed to click on the button in the top toolbar? A "Processing" box with a progress bar pops up for a second or two and then nothing. I wish I knew how to give more information than that. If I can assist in debugging in any way let me know

bennymeg, 2022-07-10:
Hi,
After it is done processing, the plugin should open the temporary directory that contain all the production files.
I assume that either the plugin failed to zip the generated file, or to open the directory in the finder.
Try running the following script:
```python
import tempfile
import webbrowser

temp_dir = tempfile.mkdtemp()
_, temp_file = tempfile.mkstemp()

try {
    print(temp_dir)
    webbrowser.open(temp_dir)
} except Exception as e:
    print(e)
}
```

if it runs try looking in the printed directory for the previously generated production files.

llange, 2022-07-19:
Hi @wraybowling , @bennymeg ; I just added a PR #4 to fix this on my macos install. Check if it works for you without regressions ?

closed 2022-07-21 by bennymeg