bennymeg/Fabrication-Toolkit #3: JLCPCB Fabrication Toolkit: how to save in the project dir?

tormyvancool, 2022-07-15:
It's not an issue but a request

I posted it here: https://forum.kicad.info/t/jlcpcb-fabrication-toolkit-how-to-save-in-the-project-dir/36684/3 they suggested me to post here

I re-write it here:
I discovered the Fabrication Toolkit JLCPCB.

Excellent! Thank you so much to all programed this tool. It makes the life way easier.

I need a help from you now.

using that tool, it saves the output to a subfolder of the TEMP folder.
Instead of manually copy and paste the content into the project folder, is it possible to set it up to save it automatically into the project folder instead of the Temp?

Thank you a zillion for this tool and for the help you will give me.

bennymeg, 2022-07-16:
Hi,
I'm happy you like the addon.

Actually, I would also prefer to save the productions files to project directory directly, but I could not find a way to get it from the KiCad api. If I would find a way to query it I would be happy to add it.

tormyvancool, 2022-07-17:
thank you Benny

i just read you into the thread on KiCAD Forum.

I hope the suggestion you received will help you to improve it and I can't wait to update it