bennymeg/Fabrication-Toolkit #13: AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetExcludeEdgeLayer'

nickoe, 2022-09-10:
I get this error in the terminal when executing it from the toolbar in pcbnew.

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/.../.local/share/kicad/6.99/3rdparty/plugins/com_github_bennymeg_JLC-Plugin-for-KiCad/thread.py", line 31, in run
    self.process_manager.generate_gerber(temp_dir)
  File "/home/.../.local/share/kicad/6.99/3rdparty/plugins/com_github_bennymeg_JLC-Plugin-for-KiCad/process.py", line 30, in generate_gerber
    plot_options.SetExcludeEdgeLayer(True)
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetExcludeEdgeLayer
```

Plugin version: 1.1.1

Kicad version info:
```
Application: KiCad PCB Editor

Version: (6.99.0-3101-g667a54ad86), release build

Libraries:
	wxWidgets 3.2
	FreeType 2.12.1
	HarfBuzz 5.1.0
	FontConfig 2.14.0
	libcurl/7.84.0 OpenSSL/1.1.1q zlib/1.2.12 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.3 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.10.0 nghttp2/1.48.0

Platform: Linux 5.19.4-arch1-1 x86_64, 64 bit, Little endian, wxGTK, i3, x11

Build Info:
	Date: Aug 28 2022 20:16:35
	wxWidgets: 3.2.0 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.79.0
	OCC: 7.5.3
	Curl: 7.84.0
	ngspice: 37
	Compiler: GCC 12.2.0 with C++ ABI 1017

Build settings:
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON

```

bennymeg, 2022-09-13:
Hi,
It seems you are using version 6.99 which typically have breaking changes that should be available in v7.
Most likely there are couple of changes the will break this plugin, and they will be addressed in a future release.

nickoe, 2022-09-14:
Just fornfuture someone, here is a note about this from upstream.

https://gitlab.com/kicad/code/kicad/-/issues/12418#note_1101084272

bennymeg, 2022-09-16:
OK, I'll close this for now, it could be ears until they reach a release candidate.

closed 2022-09-16 by bennymeg