bennymeg/Fabrication-Toolkit #14: Pick and place data is broken when uploading to JLCPCB

zapta, 2022-09-11:
When creating an SMD service order with JLCPCB, the placement data is not interpreted well by JLCPCB and all the parts are placed outside the board. 

To preproduce:
Create a JLCPCB SMD service order with the files in the attached production.zip which were generated by Kicad 6.0.7 and this plugin. When you get to the 2D/3D board view, all the components are placed away from the board as shown here https://i.imgur.com/CCbHAmx.png
[production.zip](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/9543433/production.zip)

Sanity check:
Do the same with the files in the attached easy_eda_fab.zip. These files were exported from an easy eda design and do work.
[easy_eda_fab.zip](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/9543434/easy_eda_fab.zip)

I believe that this is a recent change in JLCPCB and not specific to this plugin, but posting here such that it can be adapted to generate files compatibles with JLCPCB.

harbaum, 2022-09-12:
I have the same issue. It seems JLC PCB does not accept negative Y coordinates. They are treated like positive ones and thus the entire placement is mirrored to the top.

harbaum, 2022-09-12:
You can work around this by moving the entire PCB up, so it sits in top of the frame. Then all coordinates are positve and the parts are placed correctly.

zapta, 2022-09-12:
Moving the PCB up above the page frame fixed it. Thanks @harbaum.  BTW, Kicad and the position file have the Y axis in the opposite direction with minus at the top on Kicad and at the bottom in the position file.

@bennymeg, I suggest that the plugin will address this JLCPCB's issue, e.g. by displaying a warning if negative coordinates exist or even by adjusting the coordinates of in the production files.

zapta, 2022-09-12:
I just found that I can disable the red frame so brining the lower left corner of the board to 0,0 workaround the problem, provide 0,0 based coordinate system and looks great.

bennymeg, 2022-09-13:
@zapta I'm not sure this should be addressed.
I have ordered some PCB's couple of days ago, while the parts position looked fine in the components page it appeared outside the PCB it the orders details page.
All the PCBs came fine, so I think this is a bug in their GUI (or they can fix it manually quickly while the edit the production files in house).

zapta, 2022-09-13:
@bennymeg, yes, it's a JLCPCB bug and a recent regression as this used to work a few weeks ago.  I hope they will fix it soon.

With this bug, the normal Kicad/JLCPCB workflow which this plugin targets is broken which will probably puzzle many users of this plugin. You could add a warning if negative numbers show up with a link to this issue and the workaround. Closing this issue as is is also fine.

BTW, this plugin is a great time saver for me as I upload to JLCPCB the same design many times until all my Kicad footprints have the 'correct' offset and rotation. I run it using the icon that showed up in the PCB layout editor and having the file explorer opening automatically is a nice touch. I wish it would also upload the files automatically as EasyEda does but I don't know if it's possible without backend integration.

bennymeg, 2022-09-13:
I cannot add this feature as JLC them self and the KiCad developers forbid me to integrate this directly to their site (probably because they want people to use EasyEda instead). There is a branch with some integration code, but it needed to be finished and it would not be able to be published to the KiCad addon library.

Anyways, the PnP file is still valid (I have tested it last week). I will keep this issue open in the meantime so people could find it easily until the bug will be fixed.

zapta, 2022-09-13:
Sounds good. Thanks @bennymeg.

zapta, 2022-09-14:
I informed JLCPCB about the regression. Let's see what happens.

marekpiechut, 2022-09-18:
I have just encountered the same issue.

The simplest way to workaround the issue is to move **Drill/Place file origin** (Place menu) to the left bottom corner of the PCB and make sure to select "Use drill/place file origin" when plotting/exporting data.

zapta, 2022-09-18:
Thanks for the workaround. Looks simple indeed.

It's strange that Kicad and the CPL file use opposite directions of Y
coordinates. Personally I prefer Y+ to be up.

hildogjr, 2022-10-21:
It is legacy related to the canvas behaviour of library of WxWidgets that Pcbnew use.

zapta, 2022-10-21:
From my experience, setting Kicad footprints to have 0,0 where the JLC red
cross is results in proper x,y positioning in JLCPCB.

With these JLC x,y offset fields, modifying the kicad footprints will be
unnecessary.

hildogjr, 2022-10-22:
I have used a old plugin for while and produced some boards. I just make Y axis equal -Y: line 128 of https://github.com/hildogjr/kicad_jlcpcba/blob/master/jlcpcba_main.py

rconstanzo, 2024-03-01:
I'm not sure if this is the same error, but it just hangs on the `Processing files...` page forever for me:
[image]

I have read something about changing the rotation of the files to fix it, but I have no clue which files would need rotation or anything as nothing ever shows up.

dzid26, 2024-03-01:
> I'm not sure if this is the same error, but it just hangs on the `Processing files...` page forever for me: 
> 
> I have read something about changing the rotation of the files to fix it, but I have no clue which files would need rotation or anything as nothing ever shows up.

No, just try again

rconstanzo, 2024-03-01:
Ah right, this is a "normal" part of uploading things this way? (My first time ordering a pre-populated board)

dzid26, 2024-03-01:
> Ah right, this is a "normal" part of uploading things this way? (My first time ordering a pre-populated board)

Sometimes it just doesn't load. If it happens everytime, who knows, maybe another JLC bug. 

Btw, I tried placing grid origin somewhere on the left edge of a board to get preview in JLC cart zoom in correctly.  

And I think the original issue is not present anymore.

rconstanzo, 2024-03-01:
That's good to know, thanks for the response.

zapta, 2024-03-01:
Since it's the first time you are ordering pre-populated boards, the basic
parts are your friends ;-)  For cost considerations, try to minimize the
number of unique non basic part types. Also, if you order 5 PCBs, you can
still ask them to populate only 2, or 3, or 4.

Also, this plugin allows to add offset and rotation attributes to
footprints such that when you upload to JLCPCB you will not need to make
each time manual adjustments.

My 2c. Good luck.

rconstanzo, 2024-03-01:
In the end I managed to load it. Thankfully it's a bunch of standard 0805 parts (that I will have populated) so things showed up correctly enough. I was just worried something broke in the process of trying to load things.

[image]

I have some SMD soldering to look forward to for the other side when it arrives, but one problem at a time!

bennymeg, 2024-03-02:
> And I think the original issue is not present anymore.

Please close this issue if it was fixed by JLC.

zapta, 2024-03-02:
People report here that the bug is gone so closing.

closed 2024-03-02 by zapta