bennymeg/Fabrication-Toolkit #15: Add information to output file names

dhaillant, 2022-09-12:
Hello,
It would be great to have the output file names include some information (like InteractiveHTMLBom does for example):
the name of the project, the revision number, the date of revision, the date of creation, etc.
Thank you!

bennymeg, 2022-09-13:
Sound good.
- [Implementation Reference 1](https://github.com/openscopeproject/InteractiveHtmlBom/blob/31b20f84c81b35d5ed3bee7f189536eeff678a7f/InteractiveHtmlBom/core/ibom.py#L222-L235)
- [Implementation Reference 2](https://github.com/openscopeproject/InteractiveHtmlBom/blob/31b20f84c81b35d5ed3bee7f189536eeff678a7f/InteractiveHtmlBom/ecad/kicad.py#L687-L698)

dhaillant, 2023-03-17:
Hello, is there any progress on this enhancement?
Or is there a config file where one can set the name (and the directory) of the output files?
Thank you.

fi0, 2023-04-16:
adding board dimensions will be great too.

closed 2023-05-18 by bennymeg