bennymeg/Fabrication-Toolkit #18: Empty LCSC Part column in bom.cvs

manmountain, 2022-09-25:
The LCSC Part # column in my generated bom.cvs is empty. I have added the JLCPCB parts in Symbol Properties with field name MPN. I a lot of errors get printed to stderr when running the Fabrication Toolkit plugin .

```
Traceback (most recent call last):
  File "/home/goran/.local/share/kicad/6.0/3rdparty/plugins/com_github_bennymeg_JLC-Plugin-for-KiCad/plugin.py", line 46, in updateDisplay
    self.m_gaugeStatus.SetValue(status.data)
```

Im using:

Application: KiCad Schematic Editor

Version: 6.0.7-f9a2dced07~116~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.13

Platform: Linux 5.15.0-47-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Jul 26 2022 19:32:47
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.84.0
	ngspice: 36
	Compiler: GCC 11.2.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

and latest Fabrication Toolkit from Plugin and Content Manager.

[image]
[image]
[kicad-output.txt](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/9641415/kicad-output.txt)

phil-barrett, 2022-09-26:
I am seeing exactly the same thing.  Tried LCSC Part #, LCSC and MPN as the field names.  No joy. Installed via plugin manager at 19:30 PST. KiCad 6.0.7.  Windows 10. 64Bit.

bennymeg, 2022-09-28:
I fixed the error you are seeing, but I am not sure if this is two separate errors or not.

closed 2022-10-20 by bennymeg