bennymeg/Fabrication-Toolkit #20: Rotation in positions.csv file?

phil-barrett, 2022-09-26:
Sorry, this is a question, not a bug.  Though might be a feature request...  I searched but did not find an answer so asking here.

JLCPCB always has device rotations messed up (ok, maybe it's not their fault but often a part is not in the proper orientation in the positions.CSV file).  Is there a way to specify this other than post editing the CSV file? I know JLCPCB will correct rotations but I'd rather not rely on a tech always doing the right thing.

zapta, 2022-10-02:
I use this script to rotate kicad models. It's not perfect and sometimes I need to do minor manual fixes but it work well with JLCPCB. Each invocation rotates by 90deg so I run it 1, 2, or 3 times as needed. Sometimes I also need to move the footprint by x/y using the move-exact command in Kicad. My practice is to rotate/offset my footprint such that the land perfectly when I upload to JLCPCB. Very useful for open source projects when other people order independently for themselves.

https://github.com/zapta/ble_stepper_motor_analyzer/tree/main/kicad/tools

zapta, 2022-10-02:
>> There is not a way that I know of the to match the rotations (unfortunately).

This could be an extra footprint property. E..g.  JLCPBP_FIX =" 270/+0.1/-0.32", for rotating by 270deg and offsetting by 0.1, -0.32mm.  Just an idea.

bennymeg, 2022-10-02:
Sounds good

closed 2022-10-20 by bennymeg