bennymeg/Fabrication-Toolkit #27: Feature Suggestion: allow custom field to be placed in 'footprint' column in BOM

wuzhongjie1997, 2022-11-04:
Hi, and thanks for the great work! 

Just a quick suggestion: as the official KiCad libraries include the library name, and are often too long (ex: 'Capacitor_SMD:C_1206_3216Metric' instead of '1206' as recommended by JLC), my usual practice is to append a custom field named 'JLCFP' for all symbols and fill in the short FP name. Would be wonderful if there is some sort of mechanism to replace the 'footprint' column contents with those in some custom field for all symbols. 

Any advice appreciated!

wuzhongjie1997, 2022-11-04:
side note: the JLCFP field is also displayed on schematic for ease of checking, schematic editor automatically assigns a new color to the custom field. Whenever i make any changes to schematic, i'd open the symbol fields table to examine the footprint fields and update them as appropriate.

bennymeg, 2022-11-04:
Something around this Regex should work for resistors, capacitors, inductors, diodes, LEDs, fuses etc...
`^\w*_SMD:\w{1,4}_(\d+)_\d+Metric.*$`

bennymeg, 2023-01-06:
Added in `v2.0.0`

closed 2023-01-06 by bennymeg