bennymeg/Fabrication-Toolkit #28: Column headers/order

LPG2CV, 2022-12-17:
Probably my setup, but column headers are incorrect (not being renamed), and columns not in correct order!

I installed  the Addin. refreshed addins, and clicked on the icon that appeared. Is there something else I should be doing?

LPG2CV, 2023-01-06:
I'm still having an issue with this!
Any thoughts anyone please. :)

bennymeg, 2023-01-06:
Try the latest release a see if the problem persists

LPG2CV, 2023-01-06:
Problem still persists :(

Is there anything I can do to help resolve this?

bennymeg, 2023-01-07:
Please attach the top couple of lines of the generated BOM
Also, please provide some additional information about your system (OS, kicad version, plugin version etc.)

LPG2CV, 2023-01-07:
Thanks :)

I have the latest version of your plugin

[image]

Application: KiCad (64-bit)

Version: (6.0.7), release build

Libraries:
	wxWidgets 3.1.7
	libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jul 26 2022 02:49:38
	wxWidgets: 3.1.7 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
[positions.csv](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/10367198/positions.csv)
[bom.csv](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/10367199/bom.csv)

bennymeg, 2023-01-08:
Everything seems to work fine.
I also uploaded them to JLC to verify they work
probably just a viewing problem

LPG2CV, 2023-01-08:
It was when I uploaded to their site and was processing towards paying that the issue came up. I contacted the help desk and they advised the bom was incorrect. :)

So a little investigation on my part showed up they are being generated that way.

LPG2CV, 2023-01-08:
It must be the way I am set up :)

closed 2023-03-17 by bennymeg