bennymeg/Fabrication-Toolkit #33: JLC Rotation Offset different on back of PCB

revk, 2023-01-11:
I don't know if JLCPCB do something unusual here, but I have parts with an JLC Rotation offset which work correctly on the front of the PCB but wrong on the back. E.g. one that needed +90 on front needed -90 on back, and similarly one was 0 and 180. Does the rotation offset need adjusting when used on back of board?

As I am including the JLC Rotation Offset in my parts template it would help if I did not have to adjust this per side.

bennymeg, 2023-03-22:
I meant it would be confusing for user if so of the offset they defined will be altered.
I understand the issue, but I do not have an idea how to solve it properly.

revk, 2023-03-22:
Fair point.

Maybe a differently named offset setting that does this to match how JLC do it, which is the new setting, with the old one deprecated but retained for backward compatibility.

closed 2023-04-13 by revk