bennymeg/Fabrication-Toolkit #37: Add support for "Do Not Populate"

Shackmeister, 2023-02-20:
In Version 7, there is now a field to indiate a component is "Do Not Populate".
It would be great if this plugin would use this to leave it out of the BOM or perhaps add it but with DNP as value but remove LCSC.

bennymeg, 2023-02-22:
Sounds good,
Do you know by any chance where is the DNP variable is located in the API?

Shackmeister, 2023-02-22:
I asked one of the lead devs in a private chat and got the following answer: "It is in the attributes structure. If it is set, the attribute 'DNP' will exist"

bennymeg, 2023-02-26:
It seems that [KiCad 7](https://www.kicad.org/blog/2023/02/Version-7.0.0-Released/) uses both _exclude from BOM_ and _DNP_:
[image]
I am not sure that is the right way to go now.
Maybe it's a good idea to open to discussion with the community if _DNP_ also means _exclude from BOM_.

Shackmeister, 2023-02-26:
"Exclude from BOM" is the replacement of "Virtual Attribute"
Hence meant for logos, mounting holes, fiducials etc. Also typically set in the symbol, and not per case basis.

Shackmeister, 2023-02-26:
https://gitlab.com/kicad/code/kicad/-/issues/8908

bennymeg, 2023-02-28:
It seems that the DNP attribute is only accessible from `eeschema`.
I do not see any reference for it in the `pcbnew` API.
Can you ask at the thread you opened in the KiCad forum if that is possible? I can find it.

Shackmeister, 2023-03-01:
I'll see if I can get someone more knowledgeable than me to chime in.

zapta, 2023-03-23:
Those Kicad masking flags are still complicated but it seems that Kicad 7.0.1 added a specific flag to exclude from the position file. Does this JLC plugin support it?  I need to upgrade 7.0.0 to 7.0.1 to try it.

https://gitlab.com/kicad/code/kicad/-/issues/14332#note_1317963024

bennymeg, 2023-03-23:
After some investigation, it seems that checking the DNP flag will set exclude from position file automatically.
Which means that this feature is supported.

Shackmeister, 2023-03-23:
Will it still be added to the BOM file?

Shackmeister, 2023-03-23:
Isn't that just creating an error with JLC? 🤔
When component's are in the BOM but not in the CPL file

zapta, 2023-03-23:
Kicad needs to clean up those flags IMO. Too much room for human errors.

bennymeg, 2023-03-23:
>Isn't that just creating an error with JLC? 🤔
When component's are in the BOM but not in the CPL file

It might, but that is how KiCad design to work right now (I have read couple of post about it), I will let them decide how to address it.

Shackmeister, 2023-03-24:
Added an issue for exposing DNP support in Python
Feel free to fill in any blanks
https://gitlab.com/kicad/code/kicad/-/issues/14406

zapta, 2023-03-24:
it seems that the DNP is available in the PCB api.

BTW, it would be nice if marking a device as DNP in the schematic (and
having a large X on top of it), will automatically exclude it from the BOM
and position file. This will avoid inconsistency between the schema view
and what is actually sent to production. I think this plugin can
implement this logic.

Shackmeister, 2023-03-24:
Indeed it seems so :)
I think people and vendors like this handled in different fashions.

Lot of vendors just wants the DNP parts in the BOM but marked as DNP. Hence why it's not left out from Kicad side.
For JLC it should definitely be left out of both BOM and CPL file.

zapta, 2023-03-24:
Since this plugin is JLCPCB specific, it makes sense to pick whatever works
with JLCPCB.

My main concern is discrepancy between what is marked as DNP in the
schematic and what will actually be populated by JLCPCB.

Shackmeister, 2023-03-24:
I understand the concern, that's the main reason I'm trying to get people to support the DNP field.
Currently (as far as I can tell) the DNP marking doesn't do anything when exporting to JLC. this mean I need to manually remove them from the Exported files

zapta, 2023-03-25:
I am setting their LCSC value to DNP such that JLCPCB rejects them even
though they are listed in the BOM and position file.  Once this feature
will be implemented, I will be able to have actual LCSC ids in the DNP
parts as well, which is cleaner.

Shackmeister, 2023-03-25:
Sounds awesome :D thanks for your work!

bennymeg, 2023-03-25:
So, if I will set the LCSC Part # as `DNP` for all the part in the BOM that satisfies:
```python
    'dnp' in footprint.GetProperties().keys():
```
will be a better practice?

zapta, 2023-03-25:
I think so, because it will allow to review positively on JLCPCB what parts
were omitted.

bennymeg, 2023-03-26:
Done.
But in my testing that property was not present.

closed 2023-03-26 by bennymeg

Shackmeister, 2023-03-27:
@bennymeg Thanks for adding, How can I use the "Nightly" version of your plugin for testing.

bennymeg, 2023-03-27:
Published locally as 2.3.0.
Install it from zip (see readme for instructions).

Shackmeister, 2023-03-28:
Tested, and it works great :)

Shackmeister, 2023-05-25:
Seems like JLC is getting "smarter".
Now it will just assign the LCSC number if Value and footprint matches.
I think we might have to change Value to DNP or completely leave it out

Shackmeister, 2023-07-27:
They even assign random Partnumbers now if the LCSC partnumber is missing.
@bennymeg Would it be possible to make an option to just leave the component out of the BOM?
Currently I Manually rework my BOM files after export to remove it

bennymeg, 2023-07-27:
I'll try to think of a fix

zapta, 2023-07-27:
I think that if a part is marked as 'leave out of BOM', it shouldn't be
listed in the BOM.

Shackmeister, 2023-07-27:
@zapta 

If you refer to the field called "Exclude from bill of materials", you are correct.
Unfortunately this is not the intended use of the field.
I have described this earlier in this thread.
You can find the developer discussion here:
https://gitlab.com/kicad/code/kicad/-/issues/8908