bennymeg/Fabrication-Toolkit #39: Add an option to group identical BOMitems

zapta, 2023-02-25:
Please add an option to group components in the generated bom file as it used to be in previous options. It reduce the mental noise and clarifies what components are of the same kind.

Old style (grouped)
https://github.com/zapta/ble_stepper_motor_analyzer/blob/main/hardware/JLCPCB/version_1_4_1/bom.csv

New style (non groupedd)
https://github.com/zapta/ble_stepper_motor_analyzer/blob/main/hardware/JLCPCB/version_1_4_2/bom.csv

As a side note, just upgraded to KiCad 7.0 and notice that this plugin is now supported by the the plugin manager. This is a great improvement. Thanks.

bennymeg, 2023-02-26:
Yep, it happed because I normalized the footprint names to be more readble, and supported by the component analyzer.

It had been fixed at 2.1.0. (Currenly available localy in this repo and will be published officialy soon).

Check it out, and let me know if it solves your issue.

zapta, 2023-02-26:
Thanks Benny.   I enabled the check for plugin update and will look for the
new version.

I got spoiled by the Plugin Manager. ;-)

bennymeg, 2023-03-14:
Great,
The version was published to official plugin manager.
It should be available upon approval.

closed 2023-03-14 by bennymeg