bennymeg/Fabrication-Toolkit #41: positions.csv does not store negative values for rotation

buccaneer-jak, 2023-03-08:
About 2 months ago with version 2.0.0 (KiCad 6.0) and today with version 2.1 (KiCad 7.0) the positions file does not store the negative values only positive. 
For example:
JLCPCB Rotation Offset : 90   or
JLCPCB Rotation Offset : -90 both have the same .csv file with positive numbers even checking with Note Pad. I have also tried all the Fallback Field examples with no luck. Thus as all values are positive JLCPCB will not make any corrections unless I edit in the '-' to the .csv.

As no one else has brought this up I'm sure I must be doing something wrong but can't see where.

buccaneer-jak, 2023-03-08:
Perhaps the problem is it doesn't yet work with KiCad 7.0 as I now see that not all of the LCSC values are not in the .csv

bennymeg, 2023-03-17:
That seems odd, it should combine the footprint orientation and the offset.
So, if the component angle is 90° and the offset is 90° the positions file should have 180°.
Is that the case for you? 
As a side note, if it only works with positive numbers, for now you can use positive numbers (i.e., 270° instead of -90°)

buccaneer-jak, 2023-03-17:
I thought the problem was not working with positive numbers but after doing another test, JLC-Plugin-for-KiCad always outputs the Orientation from the footprint and not the Rotation Offset. I even tried reinstalling the plugin and KiCad with no difference. The plugin pulls the LCSC part and the other data with no problem.

UPDATE. Even the LCSC part number doesn't output an update when I change it! Perhaps the plugin is using the tstamp as a reference and at some point I changed the symbol or footprint thereby breaking the relationship?

lijon, 2023-03-31:
Did you update PCB from Schematic?

buccaneer-jak, 2023-03-31:
> Did you update PCB from Schematic?

Not quite sure why that would matter as the orientation in PCBNew happens after the update. i.e. you only really see the problem when creating the board at JLCPCB and then edit the Schematic-Symbol-Properties to add or change "JLCPCB Rotation Offset" value and then output a new position.csv file.

I will see what happens on my next board but I just used the layout in JLCPCB to orientate the parts correctly. As mentioned early outputs added the LCSC part number correctly but no longer!

lijon, 2023-03-31:
I assume the data flow is Schematic -> PCB -> JLC-Plugin -> Production Files. So if you change a symbol property in schematic, you would need to first run "Update PCB from Schematic" to get it into the PCB file. Unless the JLC-plugin reads directly from the schematic and not from the PCB? (I guess not since there's no plugin API for the schematic, and the plugin is only available from the PCB editor)

buccaneer-jak, 2023-03-31:
Well the "JLCPCB Rotation Offset" orientation is to do with how the pick and place machine see's the part in the parts holder. So really there are 2 orientations. The PCBNew one and the parts holder. 
You may be right but I don't think the all Symbol properties are passed to PCBNew just relevant ones. I will have to check when I create my next board as this one is finished and too detailed to mess with.