bennymeg/Fabrication-Toolkit #43: Place JLC in title of KiCad plugin

itskando, 2023-03-17:
[image]

itskando, 2023-03-17:
Possibly consider adding logo as well.

closed 2023-03-22 by bennymeg

itskando, 2023-03-27:
Oh. Lame. 

Thanks for providing support and for the response —

tormyvancool, 2023-07-27:
> Hi, Unfortunately, this is not permitted by both JLC PCB and KiCad developers. It seems that JLC want their users to use `easyeda` rather than KiCad. You can reach them for more information.

Well for who have ade this plugin: it's permitted: https://github.com/Bouni/kicad-jlcpcb-tools so I think it will be for you as well. Moreover the icon yu used, it's very similar (almost identical) to the one of the plugin "Replicate Layout"

itskando, 2024-02-04:
Added additional image for clarity.

itskando, 2024-02-04:
Could the title in the menu be changed more cheekily?

CLJ Plug-In For KiCad