bennymeg/Fabrication-Toolkit #54: "DNP" items getting published in BOM

xobs, 2023-04-28:
With #37 closed, this plugin shouldn't be including DNP items in either the BOM or the positions file.

However, with the latest 2.3.0 release (which reports itself as 2.2.0), items such as these are still included in the output:

[image]

bennymeg, 2023-04-30:
Are you using the latest plugin and KiCad versions?

xobs, 2023-05-01:
I'm using 7.0.2:

```
Application: KiCad PCB Editor arm64 on arm64

Version: 7.0.2-0, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 5.3.1
	FontConfig 2.14.0
	libcurl/7.87.0 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.51.0

Platform: macOS Ventura Version 13.3.1 (Build 22E261), 64 bit, Little endian, wxMac

Build Info:
	Date: Apr 17 2023 07:41:02
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.3
	Curl: 7.77.0
	ngspice: 38
	Compiler: Clang 13.0.0 with C++ ABI 1002

Build settings:
	KICAD_SPICE=ON
```

I just re-downloaded the plugin from https://github.com/bennymeg/JLC-Plugin-for-KiCad/releases/tag/2.3.0 and it still reports version `2.2.0`, but it does appear as though it's not working.

If I tick `Do not populate` then it puts a big red X through the component but still includes it in the BOM, positions, and designators files.

This may be an upstream bug, since as you noted there doesn't seem to be any way in pcbnew to know that a component is DNP.

bennymeg, 2023-05-01:
- Do not use the tag, use the release, the tag will give you the whore repo, and not the plugin.
- it might be an issue with the link between eeschema and pcbnew.

xobs, 2023-05-01:
Yep, in particular I downloaded https://github.com/bennymeg/JLC-Plugin-for-KiCad/releases/download/2.3.0/JLC-Plugin-for-KiCad-2.3.0.zip and installed it using "Install from file..."

There are three issues I've run into:

1. Kicad keeps wanting me to "upgrade" to 2.2.0. I need to remember to ignore the "Update" button because my local copy is newer,
2. The output path seems different in the newer version -- it's under ~/AppData/Local/Temp/[random-string]/ rather than alongside my Kicad project files, so I was uploading older versions of the board, and
3. It still adds it to the BOM, it just adds it on its own line with an additional `LCSC Part #` column set to `DNP`

If this BOM is fed to JLCPCB it still shows up in the components list with a tickbox next to it and it's ticked by default. Does it make sense to completely omit DNP parts from the BOM and positions files?

Shackmeister, 2023-06-13:
This issue is due to JLC linking the components together if it finds a similar Value + Footprint.
I would probably just remove the entrance to BOM + PNP file all together.
Currently I need to edit both manually in a text editor to avoid the components being on

Frank-UXV, 2023-09-03:
Hi @bennymeg Would you be ok with a PR which just removed the entry to BOM + PNP files if the DNP flag was set?

bennymeg, 2023-09-04:
@Frank-UXV there been some discussion on the matter.
I think the conclusion was that the way it is implemented now is the proper way to go.

Having said that, I'm all for providing flexibility. So if you would be able to add another screen with a checkbox: [ ] ignore DNP part, and a `RUN` button it would be awesome.

If you would still think these parts should be ignored by default, I would suggest opening a discussion and let the community decide.

bennymeg, 2023-09-21:
@Shackmeister I have added an interface with `ignore DNP` option. you can use it to remove the entry to BOM + PNP files if it is set.

Frank-UXV, 2023-09-25:
Thanks Benny! When is there a release scheduled? :)

bennymeg, 2023-09-26:
Released locally in GH.
I will hopefully submit the official PR tomorrow. (I wish I could automate this step, it's quite annoying =/).

closed 2023-09-26 by bennymeg