bennymeg/Fabrication-Toolkit #58: Fabrication fails when running from unsaved PCB

llamasoft, 2023-05-12:
https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/4a59fb0672f798ef9646803f915b3ed3bf244e96/plugins/thread.py#L26-L27

I ran into this issue when I was trying to export a panelized board for production.  KiKit's panelization utility (https://yaqwsx.github.io/KiKit/v1.3/panelization/gui/) "is designed to be run in a standalone instance of Pcbnew (not executed from a project) as the generated panel replaces the currently open board".  As a result, I was attempting to fabricate an unsaved PCB that didn't belong to a project.  It looks like the plugin still tried to do _something_ but the end result was it attempting to open `file:///production/` which didn't exist and the output files were nowhere to be found. 😅

Fisheiyy, 2023-06-11:
what version of kicad and the plugin are you using? it works just fine for me

Fisheiyy, 2023-06-15:
@bennymeg perhaps this can just be closed? I tested this and it just opened the temp_dir as it should

closed 2023-06-15 by bennymeg