bennymeg/Fabrication-Toolkit #61: Some files in gerber.zip are not recognized by JLCPCB.

zapta, 2023-05-16:
When I uploaded this 4 layers gerber zip file to JLCPCB I am getting an error messages that some of the files were not recognized.  Not sure what they are and why there are in the .zip file but would think naively that since this a JLC specific plugin, it should include only files JLC needs when it accepts a .zip file.

The main thing is that I don't know if this is normal or if somebody is missing from my order because of dropped files.

[image]
[gerber.zip](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/11483860/gerber.zip)

stormyweathers, 2023-05-17:
what version of kicad are you using?

zapta, 2023-05-17:
Application: KiCad x64 on x64

Version: 7.0.1, release build

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian,
wxMSW

Build Info:
Date: Mar 11 2023 03:41:46

bennymeg, 2023-05-18:
That's fine. I'm adding all the necessary files for production to the zip. So it acts like a full production artifact. JLC will use the files it needs for your selected setup.

zapta, 2023-05-18:
Thanks Benny.  The issue is that 1) I need to know which files are
necessary for JLCPCB and which ones are not and 2) I need to go over the
JLCPCB list and verify that all the rejected files are unnecessary.

How about having 3 'clean' files for JLCPCB and an additional zip file or a
subdirectory with all the artifacts?

Fisheiyy, 2023-06-14:
@bennymeg this should be closed because it has been addressed

zapta, 2023-06-14:
Thanks Benny. Closing.

closed 2023-06-14 by zapta

bennymeg, 2023-06-14:
Thanks @Fisheiyy,
I will publish it in the next release.