bennymeg/Fabrication-Toolkit #64: No way to override top/bottom

AlexanderBrevig, 2023-06-01:
Some footprints may have a shadow part on the other side, such as the MX keys with hotswap option.
That is, the footprint itself may be on `top` but the wanted hot swap part is to be placed on `bottom`.

AlexanderBrevig, 2023-06-15:
I found it easier to just add a PR so you can have a look :) There is nothing wrong with the footprint. 

https://github.com/daprice/keyswitches.pretty

AlexanderBrevig, 2023-06-15:
Just to clarify, it's definitely not a _very_ common thing to do, but I also have a few parts in my library that does this. It would save me (and everyone who makes keyboards using the defacto standard lib) having to manually edit the placement file.

Fisheiyy, 2023-06-15:
> This will enable the use of footprints that are placed on one side, but ha actual mounting on the other side (such as hot swaps for MX switches).
This sounds like a problem with the footprint like I was saying, where it's technically on the front but the connections are on the back, nevertheless this feature could be useful in other situations still

MajorArkwolf, 2023-11-11:
@AlexanderBrevig I was about to do a PR to add this feature as well. This can be caused by a footprint issue but it gets muddier when you have a footprint that uses both the top and the bottom of the PCB

An example of this is the Kahil V2 hot-swap switch where the hot-swap bar is soldered to the bottom but the other items are on top. Since these files are auto-generated from a script, the maintainer of that library hasn't done anything special to account for this oddity.

Personally, I would really like this feature so I can overwrite top to bottom and not have to worry about it again in the future.

AlexanderBrevig, 2023-11-13:
Do you have the ability and time to test this branch @MajorArkwolf?

MajorArkwolf, 2023-11-13:
I can test it if it’s been abandoned.

closed 2023-11-16 by bennymeg