bennymeg/Fabrication-Toolkit #65: Running the plugin creates a DRC error.

zapta, 2023-06-10:
Using Kicad 7.0.5 on Windows and plugin version 2.2.0. It's a very strange behavior:

1. In the PCB editor with all DRC's cleaned.
2. Running the Fabrication Toolkit plugin to generate production files.
3. Running the DRC verification now give an error, as if the plugin mutated something.

The issue is very consistent, and I could not see any change in the pcb file if I save it again. 

Clear DRC
[image]

After running the plugin, a single error appears:
[image]
[image]

Fisheiyy, 2023-06-11:
i doubt the plugin is doing anything, but all that error is saying is that the footprint doesnt match whats in the library, ie its not the same as what kicad knows it as

zapta, 2023-06-11:
That's correct, the question is why running the plugin creates this error
condition? It didn't exist before I run the plugin.

Fisheiyy, 2023-06-15:
Im not sure this is a problem with the plugin, let alone something that can be fixed since it doesnt do anything but read the pcb and schematic data not modify it, then produce output files for production

closed 2023-07-20 by bennymeg