bennymeg/Fabrication-Toolkit #68: Does not work for more than 6 layers

hackrmomo, 2023-06-15:
When checking the gerber files, only a maximum of 6 layers get pushed to the gerber file. A project I was working on uses 10 copper layers but not all of them get included in the gerber file.

Fisheiyy, 2023-06-15:
The latest release (2.3.0) only has support for 6 layers, but https://github.com/bennymeg/JLC-Plugin-for-KiCad/commit/4a59fb0672f798ef9646803f915b3ed3bf244e96 has added support for 8 layers, and my PR #66 has added support for all 32 possible layers. Included in this message is the latest development version of the plugin which will allow you to successfully export your files, just open the plugin manager in kicad and click "Install from file" and select the zip below.
[JLC-Plugin-for-KiCad.zip](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/11755352/JLC-Plugin-for-KiCad.zip)

bennymeg, 2023-06-15:
will be available officially in the next release. closing for now.

closed 2023-06-15 by bennymeg