bennymeg/Fabrication-Toolkit #72: Clears `production` of any files, no question asked

eflukx, 2023-06-29:
I have (had) a habit to save the files regarding production runs into a folder called _ehh..._ `production`. Containing subdirs per hardware version/revision that was/is to be produced. This includes: schematics, gerber- and position files, parts quotes, 3D/CNC files etc. 

By clicking the 'Fabrication Toolkit' icon in my PCB editor, the plugin is so 'kind' to throw all files contained in this directory away for me, with no questions asked. I get the fact that this is open source software without any guarantee and provided as-is.... but still this behavior seems somewhat 'rude' to me. Can the plugin be made to ask for a destination folder or at least ask for confirmation before deleting a whole bunch of files...? Many 🙏🏻

bennymeg, 2023-07-03:
Sorry for that,
that already fixed in the next release (#67), I will publish it this week

bennymeg, 2023-07-06:
Fixed in v2.4.0

closed 2023-07-06 by bennymeg