bennymeg/Fabrication-Toolkit #73: Please check zones are filled

obaterspok, 2023-07-02:
Hi,

When exporting fabrication outputs from KiCad it checks that all zones are filled and issue below warning if that is not the case:
[image]

It would be nice if this plugin could do the same. I just noticed because I had a board come back where I had moved a component without refilling the zone :(

-- john

bennymeg, 2023-07-03:
Sounds good, but I not sure how to check if the board if refilled after a change at the moment

Fisheiyy, 2023-07-20:
This would be nice, but im not sure how you would implement it, perhaps try enabling the auto zone refill in your settings?

PR #82 (2023-08-09): Verify zone fills
This implements automatic zonefilling before starting an export

Resolves #73
closed 2023-08-09 by bennymeg