bennymeg/Fabrication-Toolkit #76: Seems not to be working on MacOs Ventura

FrankNz, 2023-07-14:
The package loads ok via package manage and creates files for JLCPCB, however..

- No option screen(s) open
- The 'orientation' fixes from 'rotations.cf' are not applied.

Maybe the above are known problems?

MacOs Ventura (13.4.1 (a) )
Application: KiCad x86_64 on x86_64

Version: 7.0.6-0, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.13.0
	HarfBuzz 7.3.0
	FontConfig 2.14.2
	libcurl/7.88.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.51.0

Platform: macOS Ventura Version 13.4.1 (a) (Build 22F770820b), 64 bit, Little endian, wxMac

Build Info:
	Date: Jul  6 2023 22:01:07
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.82.0
	OCC: 7.7.0
	Curl: 7.77.0
	ngspice: 38
	Compiler: Clang 13.0.0 with C++ ABI 1002

Build settings:
	KICAD_SPICE=ON

bennymeg, 2023-07-20:
1. Which option screen?
2. The 'orientation' fixes from 'rotations.cf' are not applied automatically currently due to community development with manual rotation.

FrankNz, 2023-07-20:
Hi,
See my comments below..

Regards
Frank

> 
> 
> Which option screen?
Think this was my misunderstanding, the screen shots used on GitHub are actually from KiCAD to do with footprint design and placement.  

The only screen that pops up is the folder with the produced files.

> The 'orientation' fixes from 'rotations.cf' are not applied automatically currently due to community development with manual rotation.
AH, ok

closed 2023-07-21 by bennymeg