bennymeg/Fabrication-Toolkit #77: MPN not translating to the BOM

marcdraco, 2023-07-19:
This may be an old issue that's reared up or something else - but I don't have the time to look.

I see from the code and the documentation that adding a string such as MPN, LCSC Part # and so on, allows us to add the LCSC part for the picker to automate. I've been relying on this for a while now and I'll be honest, I got complacent.

That was expensive! I just ordered a load of completed boards and they came back with a 5K6 resistor in place of the 60R ferrite bead. Not the end of the world as it's easily removed of course.

Entirely my fault for not checking (I was tired, that's my excuse) and JLC were really good about it.

But to the point, no matter what string I'm feeding the BOM the actual LCSC part number never gets spat out with the BOM and JLCPCB's scripting is trying to figure out what I meant from what's left. This is the first time I've encountered this but I was on Mac previously although I was using KiCAD 7.00.

I'll fix this manually for now (the drifting components and weird rotations I can do by hand) but the MPN "bug" is a real bind as it means I've got to do a separate BOM and make sure I send the right one! :)

bennymeg, 2023-07-20:
I'm sorry you had issues with the script.
1. Does it happen only to this specific part or to all of them?
2. What key did you use, MPN? can you spot a pattern?
3. Did you remember to sync the PCB after you assign values to the schematics?

Fisheiyy, 2023-07-20:
Perhaps try using the most recent version of KiCad and the plugin?

marcdraco, 2023-07-30:
Done and done. Looks like that's fixed it. I like the new changes adding the date/time!

closed 2023-07-30 by marcdraco