bennymeg/Fabrication-Toolkit #78: Wrong components alignement on JLC

tormyvancool, 2023-07-22:
Dear

I notice that each time I upload the list to JLC, something wrogn appears regarding the components' alignemnt.
For isntance: here the diodes:

[image]

But it occurs also with Transistors and ICs.
OK JLC will corect by hands, but aside this, the files are getting wrong positioning I see.

bennymeg, 2023-07-23:
That's probably because kicad footprints and jlc footprints are not always aligned the same.
Please check that the problem does not exist when you export your pcb without the plugin.

tormyvancool, 2023-07-23:
> 

Right, good point. I tried now. Yes the result is the same. Wrong exactly as for the plugin

bennymeg, 2023-07-23:
That's way we have added option to fix the orientation for each part. we are working on implementing some automatic adjustments for some common parts, but we will never be able to fix all the misalignments that exists, it will be needed to be done manually with the options we provide.

bennymeg, 2023-07-27:
We provide configurable options.
Please read the readme file.

closed 2023-07-27 by bennymeg

blaggacao, 2026-08-28:
I wrote o LCSC support and asked them if they could add a orientation source of truth in their API, in a case of unexpected pragmatism, someone from their team might actually reach out. (or not 🤷 )