bennymeg/Fabrication-Toolkit #79: IcConfusing icon and name

tormyvancool, 2023-07-26:
Being dedicated to JLCPCB, the icon is confusion (very similar to the one of Replicate Layout" and its name is also very generic.

is it possible to have the JLCPCB Logo as icon as well as JLCPCB Fabrication as text?

This plugin does ... https://github.com/Bouni/kicad-jlcpcb-tools so there is not any impediment to adopt this way.

bennymeg, 2023-07-27:
Please read #43

closed 2023-07-27 by bennymeg

tormyvancool, 2023-07-27:
> Please read #43

I read and as I said: as it is permitted to this: https://github.com/Bouni/kicad-jlcpcb-tools s it should be to you as well .. and the current icon is too confusing when ppl uses both the plugins I mentioned, due the access of similarity

bennymeg, 2023-07-27:
It is not permitted to them. They are an unofficial plugin; you have to side load it to KiCad.
The KiCad team won't allow me to publish the plugin in the same manner.

tormyvancool, 2023-07-27:
> It is not permitted to them. They are an unofficial plugin; you have to side load it to KiCad. The KiCad team won't allow me to publish the plugin in the same manner.

Ok, strange that the one for PCBWAy and other ones, they have the "permission" and JLC not ...

bennymeg, 2023-07-27:
These plugins are developed by these company's them self... (i.e., PCBWAY)
The one who's not giving permission is JLC PCB.
Please contact JLC and KiCad for further explanation.