bennymeg/Fabrication-Toolkit #81: Error in PCB editor when clicking fabrication

SharkiBoom, 2023-08-04:
ERROR in method 'BOARD_DESIGN_SETTINGS_m_SolderMaskMargin_set', argument 2 of type 'int'

I'm not sure how to fix this and I haven't found anyone else having this issue. 
This is kicad 6.0, I will try migrating over to kicad 7.0 to see if it changes anything :)

ianog-eng, 2023-08-04:
Also having this issue with KiCad 6.0

bennymeg, 2023-08-05:
Probably because of this line: https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/ef138c8feb8282dfd7ed4f879b4f1364de1db1fd/plugins/process.py#L68
JLC PCB expects this margin.
I Checked the API and it seems KiCad excepts an `int` ([here](https://docs.kicad.org/doxygen-python-6.0/pcbnew_8py_source.html#l09199)).
I haven't encountered this issue before, probably the units KiCad want are not millimeters, I am trying to figure this out.

JHay0112, 2023-08-11:
I have seen this issue in KiCad 6.0 as well. A quick Google says that KiCad internally stores measurements as nanometres (https://docs.kicad.org/6.0/en/pcbnew/pcbnew.html) so perhaps this is what the API is expecting?

bennymeg, 2023-08-14:
OK, I'll try to update it to 50000 nanometers.

bennymeg, 2023-08-19:
Added a fix in 3.0.2 (prerelease).
you can try it manually.

closed 2023-09-26 by bennymeg