bennymeg/Fabrication-Toolkit #85: JLCPCB Plugin not running to completion

FrankNz, 2023-08-18:
JLC Plugin
Running on Mac. (13.5)
The Plugin creates all the files in the temp directory, but, failed when trying to rename the .ZIP file around line 100 in the thread.py file. 
It seems to construct a new name which would place the renamed file inside a folder that it is in! The rename process wants the renamed file in the same 'location'. I'm telling you nothing new here.
I know about '0' Python to be of any help fixing this. Commenting out the 'offending' os.renaming line allows the program to run to completion. The zip file in the production directory has a 'temp' filename. 

UPDATE
I think the problem is that I had a '/' in the Title of the PCB. Removing the '/' allows your program to run properly!

bennymeg, 2023-08-18:
Nice catch, I'll normalize the file name.

closed 2023-08-18 by bennymeg

FrankNz, 2023-08-18:
Hi Benny,
Thanks for the quick fix.
If I uninstall/install the plugin it will pickup the fix?
This will fix the ‘glitch’ for windows users also?
Regards
Frank

bennymeg, 2023-08-19:
No, I need to create a new release first.
Then, you will have to install in manually from Github (see Readme.md) because KiCad official repository takes a lot of time to publish a new release.

FrankNz, 2023-08-20:
Looks good, Thankyou