bennymeg/Fabrication-Toolkit #86: Add prompt to re-fill pours if the pours are out of date

DeflateAwning, 2023-08-24:
Just did an off and ordered a PCB with out-of-date pours. Some of the traces are shorted to the ground pour, because I moved the traces into the pour last minute and then didn't re-pour. 

My bad.

Would've been nice if it prompted with a warning saying that the pours are out-of-date. This is a feature in the regular Gerber exporter, iirc.

If this feature was already added, I wasn't on a late enough version (but am not sure what version I was on).

obaterspok, 2023-08-26:
This has been fixed to always perform a re-fill prior to an export

Duplicate of #73

bennymeg, 2023-08-26:
I'm sorry to hear that.
As @obaterspok mentioned, it had been fixed (by him [thanks for the contribution]) in v3.0.2.

closed 2023-08-26 by bennymeg