bennymeg/Fabrication-Toolkit #87: PLOT_CONTROLLER error

21km43, 2023-08-25:
When I update the tool version from 2.4.0 to 3.0.0 and run it, I get the following error...
[image]

bennymeg, 2023-08-26:
This seems to be a KiCad issue.
Please update KiCad.

makuser, 2023-09-07:
> This seems to be a KiCad issue.

What are you even talking about... It's not a KiCad issue, it's a You issue.
Your plugin is trying to call a future KiCad function for v-cuts, that does not exist in KiCad 6, yet your plugin claims compatibility with KiCad 6.

There's no `bool PlotLayers()` in KiCad 6: https://gitlab.com/kicad/code/kicad/-/blob/6.0/pcbnew/plotcontroller.h

The plugin should not try to use functions that are not there.

makuser, 2023-09-08:
> When I update the tool version from 2.4.0 to 3.0.0 and run it, I get the following error...

If you can not "update KiCad" from 6 to 7, you can open one of the plugin code files in a plain text editor (no Word or so, it must handle raw plain text) or code editor if you happen to have some installed and patch it yourself, to be able to continue working with this plugin as you should.

Edit the process.py file as follows:
```patch
diff --git a/plugins/process.py b/plugins/process.py
index af3b8c1..46bd13f 100644
--- a/plugins/process.py
+++ b/plugins/process.py
@@ -113,11 +113,7 @@ class ProcessManager:
                     layer_info[0],
                     pcbnew.PLOT_FORMAT_GERBER,
                     layer_info[2])
-                seq = pcbnew.LSEQ()
-                seq.push_back(layer_info[1])
-                if layer_info[1] == pcbnew.Edge_Cuts :
-                    seq.push_back(pcbnew.User_1)
-                plot_controller.PlotLayers(seq)
+                plot_controller.PlotLayer()
 
         plot_controller.ClosePlot()
 ```

thermseekr, 2023-09-08:
Thank you @makuser! I was about to open another issue saying there was a problem running the plugin on a Mac, as on my PC it works, when I decided to check the closed issues. So probably on the PC I have a newer Kicad release. I just cannot update Kicad on the Mac now, I'm in a remote place with a very weak internet connection. 

@bennymeg thank you for the plugin.

bennymeg, 2023-09-12:
> What are you even talking about... It's not a KiCad issue, it's a You issue. 

Thank you for your kind words...
What I meant is that it only missing on KiCad v7.0.0 and available, on all the versions afterwords, hence the KiCad issue deduction.

It seems that I've missed it whilst merging PR #71.
I wasn't aware that `PlotLayers` is missing in KiCad 6, I'll try to fix it by removing v-cut support for v6.

closed 2023-09-12 by bennymeg