bennymeg/Fabrication-Toolkit #92: output_path characters not normalized

Fisheiyy, 2023-09-19:
when exporting files with a odd characters such as ^ it will pass to the name of the output folder and the name of the gerber archive which causes issues when uploading to JLC for production

bennymeg, 2023-09-19:
This should be handled.
Please provide an example.

Fisheiyy, 2023-09-21:
> This should be handled. Please provide an example.

I havent tested with any other special characters, but the ^ one gets past the current normalization that is happening so it causes an error when uploaded to JLC unless I manually remove the ^
[image]

bennymeg, 2023-09-21:
I just saw you are using v3.0.0
This was fixed at v3.0.2 (currently in prerelease).

- (#85)

closed 2023-09-21 by bennymeg