bennymeg/Fabrication-Toolkit #93: Plugin fails to start with error in plugin.py line 12, "The wx.App object must be created first!"

monohydroxy, 2023-09-21:
After installing the plug-in and restarting KiCAD, we I push the Fabrication Toolkit button it immediately fails with the error below:
    The plugin fails to start with error in plugin.py line 12, "The wx.App object must be created first!"

I'm running Ubuntu 22.04 and KiCAD 6.0.2+dfsg-1,release build.  The plugin is version 3.0.0 and was installed from within KiCAD from the repository.

Am I doing something wrong or do I have a mismatch of versions?

monohydroxy, 2023-09-21:
Thank you for the suggestion.  Unfortunately it did not work.  I added some new text to the PCB and saved, but I still got the same error.  I also updated to 6.0.11 and the problem persists.

In a google search I found this strange website with a suggestion to add the line 'self.app = wx.PySimpleApp()' in the __init__ function.  I tried that suggestion,and it gave me a different error and then crashed.

Is the plugin supposed to create the ws app or does KiCad create it?  Is this a new feature of the wx framework?  I'm not familiar with it.

Strange website:
  https://techoverflow.net/2023/03/01/how-to-fix-kicad-jlcpcb-fabrication-plugin-error-the-wx-app-object-must-be-created-first/
New error:
  in method 'BOARD_DESIGN_SETTINGS_m_SolderMaskMargin_set', argument 2 of type 'int'

bennymeg, 2023-09-21:
Adding 'self.app = wx.PySimpleApp()' cause weird issues.
The second issue was fixed in 3.0.2.

monohydroxy, 2023-09-21:
I think this is related to:  https://forum.kicad.info/t/solved-nightly-6-99-broken-scripting-console-linux-ubuntu-22-10-22-04/39302
My scripting console doesn't work as well.  It seems to be a mismatch between wxPython, wxWidgets and KiCad version assumptions.

monohydroxy, 2023-09-21:
The plugin works without issues with KiCad 7.0.7 installed from a Snap.  Closing this issue since I'll move my project to to version 7 soon and I don't think it is a plugin bug.

closed 2023-09-21 by monohydroxy