bennymeg/Fabrication-Toolkit #94: Caps and Resistors "LCSC Part #" not included in BOM

Elithraldor, 2023-10-04:
I'm on Kicad7 - if I fill out the LCSC Part # field and generate the production output, the caps and res LCSC Part # field is empty. Is it supposed to be this way?

Elithraldor, 2023-10-04:
Let me know if you need more (are there any logs printed somewhere?):

```
Application: KiCad x64 on x64

Version: 7.0.7, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Aug 14 2023 02:42:39
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.7.1
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON
```

Fabrication Toolkit installed through the Plugin And Content Manager is version 3.1.0.

bennymeg, 2023-11-16:
Did you sync the schematic with the board before generating the bom?

igoro00, 2023-12-10:
> Did you sync the schematic with the board before generating the bom?

I had the same problem. This solved it.

closed 2023-12-11 by bennymeg