bennymeg/Fabrication-Toolkit #98: Rotation and Layer are reversed

GW8RDI, 2023-11-02:
Trying to upload the files to JLCPCB but it says the CPL file isn't correct. Checking the JLCPCB sample of the format the layer and rotation columns are reversed.

bennymeg, 2023-11-02:
Reverse them manually an check if it fix the problem.

GW8RDI, 2023-11-02:
Do you think it's a KiCAD maintenance issue with the 3D models, some models are okay, others rotated almost at random.

bennymeg, 2023-11-13:
Are you talking about the columns or the rotation?
The column order should not matter.
The rotation of KiCad components and JLC components do not always match and needed to be manually adjusted.
We are working to improve the default rotations for some common components.

closed 2023-11-13 by bennymeg