bennymeg/Fabrication-Toolkit #102: JLCPCB internal server error when BOM line is too long

romain145, 2023-11-15:
When uploading a matrix of LEDs onto JLCPCB the website keeps showing an "Internal server error" when processing the BOM and CPL files for a PCBA that has more than 350 LEDs.
After discussing with an agent, they fixed the BOM by splitting the LED references into 2 lines.
It appears that their backend has a maximum of references they can process on each line. I do not know the exact number.

As a side note, they also renamed the column "Value" to "Comment" to match their template:
https://jlcpcb.com/parts/bom-tool/?url=https%3A%2F%2Fjlcpcb.com%2Fparts%2Fbom-tool%2F&_t=1700044499402

bennymeg, 2023-11-16:
The max BOM line should be 256 items:
https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/8db1a4d446336aa4d34d0a66cc8adbae25e3a9d9/plugins/config.py#L11

Are you sure you are using the latest release?

romain145, 2023-11-16:
[image]

The config.py doesn't have the bomRowLimit entry, even after reinstalling the plugin:

```
import pcbnew  # type: ignore

baseUrl = 'https://www.jlcpcb.com'

netlistFileName = 'netlist.ipc'
designatorsFileName = 'designators.csv'
placementFileName = 'positions.csv'
bomFileName = 'bom.csv'
gerberArchiveName = 'gerbers.zip'
outputFolder = 'production'

plotPlan = [
```

I added the line manually to config.py, closed and reopened KiCAD but all the LEDs are still on one line.

bennymeg, 2023-11-16:
My bad, I didn't released it yet.
Will be fixed soon.

closed 2023-11-16 by bennymeg