bennymeg/Fabrication-Toolkit #106: Add support for offset in the database.

Shackmeister, 2023-11-29:
Currently the database only support rotational offsets.
Including X and Y would greatly improve the ease of use

bennymeg, 2023-11-29:
You can use 'JLCPCB Position Offset' for that.

closed 2023-11-29 by bennymeg

Frank-UXV, 2023-11-29:
Thats a field inside the schematic correct?

Shackmeister, 2023-11-29:
I currently have 100+ design , so being able to have the database externally is definitely nicer.
This also makes it easier to make a combined effort

bennymeg, 2023-11-29:
> Thats a field inside the schematic correct?

Yes

> I currently have 100+ design , so being able to have the database externally is definitely nicer.
This also makes it easier to make a combined effort

I'm not sure I understand, please explain further.
Keep in mind the there is an offset database that will fix most of the common components offsets (if that hat you meant).
It will take effect in v4 since it is a breaking change.

Shackmeister, 2023-11-29:
`I'm not sure I understand, please explain further.
Keep in mind the there is an offset database that will fix most of the common components offsets (if that hat you meant).
It will take effect in v4 since it is a breaking change.`

My point was that I would prefer not to add the schematic field to all my kicad projects. Preferably a database outside the kicad project would be highly preferable.

Is this the database you are refering to? https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/master/plugins/rotations.cf
If so, my original point was it only supports rotational offset, and not X/Y offset.
 If we could add something like 
`^VSSOP-10_-         270 -1.0 3.5`
to tell it to be 270 degrees, X offset = -1mm and Y offset = 3.5mm. it would be all I need :)

bennymeg, 2023-11-29:
I do think the X/Y offset is the same for all the packages (i.e. I think the are many VSSOP-10 variants).
So, what works for rotation wight not work for translations.
This need to be verified, but currently I am not planning to explore it (it seems like a rabbit hole).

Shackmeister, 2023-11-29:
The VSSOP was only an example.
I think there are plenty of cases which will not need the offset.
Connectors however are very prone to need offsets as well as rotations

Shackmeister, 2023-12-20:
@bennymeg I have been playing around with the rotations.cf, but no matter what I change, I cannot get a rotations added in the output file. Can you please have a look if something is broken?
For instance an SOT-23 footprint will always get 0 deg rotation added, eventhough it's set to 180.

Shackmeister, 2023-12-20:
I see it's not hooked up on line 192¨here:
https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/master/plugins/process.py
Trying to enable it gave me:
" 'FOOTPRINT' object has no attribute 'split'

bennymeg, 2023-12-20:
You can try testing PR #96.
It would be included in v4.

Shackmeister, 2023-12-20:
Thanks! I'll give it a try. Any idea when V4 will come out? :)