bennymeg/Fabrication-Toolkit #108: Optionally Tent Vias

Zerick, 2024-01-04:
Is there a way to select tented vs untented vias?
If not, I would like one.

Thanks!

Zerick, 2024-01-12:
I did find a work around. If one skips step 3 this tool will default to the last style generated rather than the setting in the Gerber generation wizard.

1 One must check that vias are either tented or not as you wish on Gerber Generation!!!
2 If you want the OTHER option, set it in the gerber generation wizard
3 then you MUST manually generate gerbers ONE time for KiCAD to save this setting <-this is so strange
4 now verify that the plugin is generating gerbers correctly as you wish them (vias tented or not)
5 one only need check the vias on the mask layers to confirm

dzid26, 2024-01-13:
I think you can use this, simply:
[image]

closed 2024-03-12 by bennymeg