bennymeg/Fabrication-Toolkit #112: KiCad 8.0.0-rc1 API changed

akakoziqi, 2024-01-19:
In the 8.0.0-rc1 version of KiCad, there is an API change.

When I use this plugin in KiCad 8.0.0-rc1, there is an error:
[image]

The reason is footprint object in process.py has no ```HasProperty``` and ```GetProperty``` function. 
I replace the ```HasProperty``` function by ```HasFieldByName``` function, and replace the ```GetProperty``` function by ```GetFieldByName``` function.

Then it seems worked.

But I don't know how to make if capable for the previous version of Kicad. :)

boromyr, 2024-01-19:
@akakoziqi it worked thanks

bennymeg, 2024-01-20:
Thanks for letting us know, a solution had already implemented here: [#109].
Will be released in the next major release (v4).

closed 2024-01-20 by bennymeg