bennymeg/Fabrication-Toolkit #118: Won´t export LCSC Part #

catalinalb, 2024-02-20:
Good day,

I have KiCad 7.0.7 (also try on 7.10 with same results) I did add the LCSC Part # for the first component and try the plugin and it did exported the number. Of course after finishing all the componets in the schematic I got an empty cell. No LCSC numbers.... 

What have I done wrong? 

[image]
[image]

bennymeg, 2024-02-20:
Did you synchronized the schematic with the board after you defined all the MPN's?

catalinalb, 2024-02-20:
Do you mean to annotate ?
yes I did that....

now it won´t generate the gerber archive .... :(

bennymeg, 2024-02-20:
No, I mean update, __F8__

catalinalb, 2024-02-20:
Indeed, it works now! Thank you so much for this ! 
I did know that I had to annotate but didn´t know about the F8 update.

It works now! 

Ticket can be closed!

closed 2024-02-20 by bennymeg