bennymeg/Fabrication-Toolkit #121: Some parts do not seem to obey the JLCPCB Rotation Offset..

frohro, 2024-02-22:
I cannot find any typos, but some parts do not rotate properly.  Others do just fine with the addition of the field JLCPCB Rotation Offfset and the values in degrees.  I'm not sure why.  Two of my ICs on the lower right of my board do not change orientation with any value in the rotation offset field of the schematic editor when I look at the JLCPCB preview.  The Chat helpers don't seem to put much stock in the preview, but changing the rotation by hand editing the placement file rotates things, so I don't think that is the problem.

harbaum, 2024-02-27:
You could take a look at the generated positions.csv. It's an ASCII file and you should see the rotation values arriving there in the 4th column.

bennymeg, 2024-02-27:
press F8 to update the board properties from schematic.

closed 2024-02-27 by bennymeg

tavdog, 2024-03-05:
This is happening to me too.  
[image]

harbaum, 2024-03-06:
Did you try to debug or fix it as described in previous replies? Do the rotation values show up in the location file? Did you update via f8?

tavdog, 2024-03-06:
I managed to git it working by using the other jlcpcb plugin which allowed me to manually set the rotation for the parts in question.