bennymeg/Fabrication-Toolkit #122: `FOOTPRINT` object has no attribute `HasProperty`

misteraddons, 2024-02-23:
[image]

Freshly installed on KiCad 8 (perhaps it's not yet supported). I haven't used this plugin before but installed it using the official plugin manager.

revk, 2024-02-24:
Doing the same for me. Obviously an 8.0 issue.

The extra option screen is a nuisance, can we lose that somehow?

Anything we can do to help?

tcurdt, 2024-02-24:
Seems like the API has changed.

  https://github.com/bennymeg/JLC-Plugin-for-KiCad/commit/8dc0e8a440d2cefc9c1b96be43f5494eeb7fa9a3

and the required change was already merged.

  https://github.com/bennymeg/JLC-Plugin-for-KiCad/pull/109

Maybe we are just not getting the most recent version?

tcurdt, 2024-02-24:
Yes, 3.1.0 does not have the fix yet. So it needs a new release.

https://github.com/bennymeg/JLC-Plugin-for-KiCad/compare/3.1.0...master

Work around could be to install master for now.

revk, 2024-02-24:
I'll try that.

revk, 2024-02-24:
I can confirm the master works, is there any way to remove the asking the option at the start?  I can't see quite what to comment out to do that.

tcurdt, 2024-02-24:
> I can confirm the master works, is there any way to remove the asking the option at the start? I can't see quite what to comment out to do that.

I assume you would need to replace [this here](https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/master/plugins/plugin.py#L94) with `ProcessThread(self, dict())`

But that's just from briefly looking at the code.

revk, 2024-02-24:
Yeh, I got as far as

```
        options = dict()
        options[EXCLUDE_DNP_OPT] = False;
        ProcessThread(self, options)
```

And no joy.

revk, 2024-02-24:
Unrelated, but a part that previously had a position offset of 5,0 now needs 0,2.61

I have no clue why!

tcurdt, 2024-02-24:
> Yeh, I got as far as
> 
> ```
>         options = dict()
>         options[EXCLUDE_DNP_OPT] = False;
>         ProcessThread(self, options)
> ```
> 
> And no joy.

Ah... seems it does need the dialog.

Check out an older version:

https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/2.3.0/plugins/plugin.py

bennymeg, 2024-02-24:
> The extra option screen is a nuisance, can we lose that somehow?

I have added it due to a request to add export options (Currently for dnp components). 

> Unrelated, but a part that previously had a position offset of 5,0 now needs 0,2.61
I have no clue why!

The positioning algorithm had been improved substantially, and a few breaking changes has been introduced to v4 of the plugin.
Read more in the upcoming official release.

---

Duplicate of (#109, #112, #114, #119).

closed 2024-02-24 by bennymeg