bennymeg/Fabrication-Toolkit #132: Kicad 8.99 does crash

robouden, 2024-03-01:
With Application: KiCad Version: 8.99.0-unknown-cd27f801f6~177~ubuntu22.04.1

When starting the plugin, it crashes KiCad.
PCWway plugin has similar issue with KiCad 8.99, and they have revolved. Maybe if the team from the JLCPCB plugin fabrication tool kit can look how the other team ([Link](https://github.com/pcbway/PCBWay-Plug-in-for-Kicad)) implemented some simple changes with their plugin for KiCad V8.99, that would help, I hope. 

I love to get this plugin working like the PCBWay plugin work snow with KiCad.

Warm Regards,
Rob Oudendijk

bennymeg, 2024-03-01:
I am not planning to support nightly versions of KiCad.

I have checked the link and did not see any major change.
I have added a patch to hopefully fix the issue, but if it would not, I will address it once the stable release of KiCad 9 will be released.

robouden, 2024-03-01:
I downloaded the master zip file and tried to add it as the plugin. Tried to install it but get install error.
What to do?

[image]

bennymeg, 2024-03-01:
Download the 4.1.0 release and replace (in the zip) the utils.py file with the new one from master

danielchisholm, 2024-03-01:
> Download the 4.1.0 release and replace (in the zip) the utils.py file with the new one from master

Thanks Benny.  4.1.0 still crashed, but manually putting in utils.py from 0917e9f worked for me.

bennymeg, 2024-03-02:
Fixed in 4.2.0

closed 2024-03-02 by bennymeg