bennymeg/Fabrication-Toolkit #133: Index out of range

xzf16, 2024-03-04:
When I using fabrication toolkit, I got an error message: Index out of range.
[image]

Kicad version is:
Application: KiCad PCB Editor x64 on x64

Version: 8.0.0, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb 23 2024 02:24:15
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:

Fabarication toolkit version is:4.2.0

dzid26, 2024-03-05:
Strange, I also am getting this now on a simplest board possible, while it still works on a super complicated one. 
I have a feeling it's related to layer I imported from altium. 
[ford_q3.zip](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/14500137/ford_q3.zip)

xzf16, 2024-03-06:
> Strange, I also am getting this now on a simplest board possible, while it still works on a super complicated one. I have a feeling it's related to layer I imported from altium. [ford_q3.zip](https://github.com/bennymeg/JLC-Plugin-for-KiCad/files/14500137/ford_q3.zip)

The board I have was designed by KiCAD originally, so I don't think it's related to importing from altium issue.

And the board was my companies product, so I can't upload it here.

Tjookum, 2024-03-07:
I have the exact same issue since updating to kicad 8.

bennymeg, 2024-03-07:
Can you find what's common on the PCB's that does not work?

dzid26, 2024-03-07:
> Same Problem here. Can´t Upload PCB, because it´s Product from my Company

I know the problem. 

If you need to fix it immediately, find graphics that are set as "unspecified" (for example graphics) and Exclude from Position files. 
[image]

PR #138 (2024-03-12): Footprints without pads fix #133
Graphics can be stored as footprints - usually with `Unspecified` footprint type. That triggered footprint position calculation based on [pads bounding box centroid](https://github.com/bennymeg/JLC-Plugin-for-KiCad/pull/117) (intended weird footprints imported from other programs, not for graphics).  Graphics don't have pads which was causing Index out of range error. 

Changes:
- skipping footprint without pads - fixes #133
- small refactor (put all skipping conditions in `skip_footprint` var)
- added extra comments (because I was getting confused about my own [... cut]
closed 2024-03-12 by bennymeg