bennymeg/Fabrication-Toolkit #135: New update screws up things!

AndyW999, 2024-03-05:
I was quite happily using your last version and all was well, I use the EDA footprints from the JLCPCB EDA, export them, convert them to KiCAD with 

https://wokwi.com/tools/easyeda2kicad

And all the parts are in the correct position and the correct rotation in the previous version.

You last update seems to have rotated some chips that were previously OK and offset some other parts that were correctly placed.

Thanks

Andy.

AndyW999, 2024-03-05:
Version 3.1.0 is OK, all version 4 presents this problem.

bennymeg, 2024-03-06:
v4 has some breaking changes.
Please read the release docs.
I might add an option to choose whether to add the automatic translation in the future.

closed 2024-03-06 by bennymeg

AndyW999, 2024-03-06:
So I will keep using V3 then...

AndyW999, 2024-04-18:
'I might add an option to choose whether to add the automatic translation in the future.'

What is this and will it un-break it from before as it is really inconvenient as I have quite a large library which used to work...

And now of course V3 will not work with KiCad 8 as there is an error - 'FOOTPRINT' object has no attribute 'HasProperty'