bennymeg/Fabrication-Toolkit #144: Additional layers: can't select user-defined layers

TheColorman, 2024-03-28:
When exporting using Fabrication Toolkit, we have the ability to write the names of additional layers to be exported, but this doesn't include any user defined layers, e.g. `User.[0-9]`. It seems the layers you can use are hardcoded here: https://github.com/bennymeg/Fabrication-Toolkit/blob/ab0b336ed9207d7636f2f6a2a7d2d4847c0c3b5a/plugins/config.py#L15-L56

I suggest using a way to dynamically get the enabled layers, such as
```py
layers = []
i = pcbnew.PCBNEW_LAYER_ID_START
while i < pcbnew.PCBNEW_LAYER_ID_START + pcbnew.PCB_LAYER_ID_COUNT:
    layer_std_name = pcbnew.BOARD.GetStandardLayerName(i)
    layer_name = pcbnew.BOARD.GetLayerName(board, i)

    layers.append((layer_std_name, i, layer_name))
    i += 1
```
Adapted from [Board2Pdf](https://gitlab.com/dennevi/Board2Pdf/-/blob/b4cbd95d2bacdd83655dbb5192b62a33050ba817/dialog/settings_dialog.py#L323-335).

Additionally, the additional layers input box seems to support a list of comma-separated layer names.
https://github.com/bennymeg/Fabrication-Toolkit/blob/ab0b336ed9207d7636f2f6a2a7d2d4847c0c3b5a/plugins/process.py#L72
This isn't very clear to the user, especially as the autocomplete doesn't acknowledge attempts at adding more than one layer, so I feel like some type hint would be useful.

I'm working on a PR to address these issues, but I'd like to hear some feedback about it here first.

bennymeg, 2024-03-29:
Sounds good.
Regarding the autocomplete, I agree, I could not make it work properly for more than one layer, and the UI is quite limited and could not make it look nice, so I opted to ignore it.
If you could add it properly, I'm all for it.

TheColorman, 2024-04-09:
I've opened PR #152 that makes the autocomplete list dynamically generated. I tried to play around with having a list of layers underneath the text box that you could add/remove from, but couldn't get anything that looked nice, so for now I've just updated the textbox hint.

closed 2024-04-09 by bennymeg