bennymeg/Fabrication-Toolkit #148: BOM generation broken

TheColorman, 2024-04-03:
BOM generation was broken in 88ae87bf48d12ef76094094ce9dfae731e9aca18 (not stable release). After this commit, BOM is only generated if "Exclude DNP components" is checked.

I'm not 100% sure why as I have a bit of a hard time wrapping my head around the KiCad api, but thought I'd make an issue regardless.

bennymeg, 2024-04-16:
Could you please test again with the latest release.

CreativeRobotics, 2024-04-18:
I just installed the latest release and now BOM and position generation is broken when "exclude DNP.." is selected.

fh-nbt, 2024-04-19:
Can confirm the same on my machine and setup. No BOM or position files are being generated.
Win11 
Kicad 8.1 
Fabrication-Toolkit 4.30

bennymeg, 2024-04-20:
Does some part get excluded or the entire files do not get generated?
Could you please test again with the source code (no release yet).

bennymeg, 2024-04-24:
I traced the bug to here: #128 
The added condition seems to misbehave, some more exploration needed.

msalau, 2024-04-25:
@bennymeg 
I traced the issue to the commit https://github.com/bennymeg/Fabrication-Toolkit/commit/88ae87bf48d12ef76094094ce9dfae731e9aca18

```
            skip_footprint = ((footprint.GetAttributes() & pcbnew.FP_EXCLUDE_FROM_POS_FILES)
                                or footprint.GetPadCount() == 0
                                or exclude_dnp
                                    and (footprint_has_field(footprint, 'dnp')
                                        or (footprint.GetValue().upper() == 'DNP')
                                        or getattr(footprint, 'IsDNP', bool))
            )
```

The issues I see:
1. `getattr(footprint, 'IsDNP', bool)` is not called anymore. It was `getattr(footprint, 'IsDNP', bool)()` but became `getattr(footprint, 'IsDNP', bool)` and since it is a function reference and is not `None` - it evaluates to `True`
2. not enough parenthesis

How to fix:
```
            skip_footprint = ((footprint.GetAttributes() & pcbnew.FP_EXCLUDE_FROM_POS_FILES)
                                or footprint.GetPadCount() == 0
                                or (exclude_dnp
                                    and ((footprint_has_field(footprint, 'dnp')
                                        or (footprint.GetValue().upper() == 'DNP')
                                        or getattr(footprint, 'IsDNP', bool)())))
            )
```

I've tested the fix on KiCAD v8.01 with the plugin 4.3.0 + fixes. 

In the master branch the fix should look like below:
https://github.com/bennymeg/Fabrication-Toolkit/blob/master/plugins/process.py#L162
```
            is_dnp = (footprint_has_field(footprint, 'dnp') 
                      or (footprint.GetValue().upper() == 'DNP') 
                      or getattr(footprint, 'IsDNP', bool)())
```
I.e. add parenthesis to call the function returned by `getattr()`

bennymeg, 2024-04-25:
Thank you @msalau, I'll fix it.

closed 2024-04-25 by bennymeg

bennymeg, 2024-04-25:
Fixed in v4.4.1

commit 05e10eba (2024-04-25): fix: dnp attribute invocation, closes #148