bennymeg/Fabrication-Toolkit #149: User layers being exported into edge cuts gerber

partlyhuman, 2024-04-05:
Hello!

Informational graphics from my User.1 layer are being exported into the edge cuts gerber when using the fabrication toolkit, causing an error in fabrication(!) I'm not sure how to change this behaivour. In the board stackup, User.1 is defined as a "User defined layer". This issue doesn't occur when using export gerbers normally.
[Archive.zip](https://github.com/bennymeg/Fabrication-Toolkit/files/14889907/Archive.zip)

From:
https://github.com/partlyhuman/loopycart/tree/main/pcb/flash-32mbit
Note I haven't pushed the change, but this user layer was called "User.FrontReference". I renamed it to "User.1" hoping that this would fix the issue, but it did not.

Please let me know if I can provide more information. Thank you!

bennymeg, 2024-04-16:
This feature was requested by the community in order to allow adding additional data for i.e. edge cuts.

partlyhuman, 2024-04-16:
> This feature was requested by the community in order to allow adding additional data for i.e. edge cuts.

Can you explain a little how it's determined which extra graphics are exported into which layer, so I can avoid this happening unintentionally?

bennymeg, 2024-04-20:
only the User.1 is combined with the edge cut layer for penalization and such.
You are right, I will add an option for this feature.
In your experience, which layer is commonly used for this use.