bennymeg/Fabrication-Toolkit #151: Copper layers are re-filled on export

victor1115, 2024-04-08:
Hello,

In my use case I need to keep the filling of the copper as is.

When I export there is an automatic auto-fill for the copper layers that can cause to interrupt GND tracks.

Before Fabrication Toolkit export:
[image]

After Fabrication Toolkit export:
[image]

Can you unset/remove the filling option?
Or make that optional using a check button like the "Exclude DNP components"?

[image]

Thank you!

victor1115, 2024-04-08:
OS: Windows 10 Pro
Version: 22H2

KiCad Version: 7.0.10, release build

Fabrication Toolkit Version: 4.1.0

bennymeg, 2024-04-16:
This feature was requested by the community.
I would be happy to accept a PR the toggle it with an option.