bennymeg/Fabrication-Toolkit #155: Compatibility on Mac 14.4.1, KiCad 8.0.1 & Fabrication Toolkit V4.1.0

FrankNz, 2024-04-16:
### Discussed in https://github.com/bennymeg/Fabrication-Toolkit/discussions/154

<div type='discussions-op-text'>

<sup>Originally posted by **FrankNz** April 15, 2024</sup>
Hello, 
The Fab-Toolkit runs on the above combination but does not present the config menus as displayed on Github. The only window that opens is the one below. I have used this S/W before on earlier Mac/KiCad combinations and it worked ok then.
Clicking 'Generate' does generate a set of files. 
Maybe I'm doing something wrong?
[image]
</div>

bennymeg, 2024-04-20:
what other screens? all the other screens in the readme are from the schematic editor.
It seems that you do not have the latest version.

FrankNz, 2024-04-20:
Thank you for the reply.
You are correct, it seems I need more coffee and to read more carefully. 
I have now updated to the latest release. Thank you

closed 2024-04-20 by FrankNz