bennymeg/Fabrication-Toolkit #159: exclude component from bom only if labled "Exclude from bill of material" but not if "Exclude from board"

CodeLima, 2024-04-20:
This is a very great plugin! Thank you very much for your effort you put into this! 

There is one enhancement which would be very nice to have in this plugn. JLC offers to populate both ends of a connector. For this is would be necessary to include parts in the BOM who are not populated on the board. As reading from the usage instruction, this plugin does remove items from the BOM it either "Exclude from bill of material" or "Exclude from board" is set. If this could be limited to only exclude it if "Exclude from bill of material" is set, the case mentioned here would be possible as well.

bennymeg, 2024-04-20:
Thanks.
Sound logical, I did not get a chance to use this service yet, probably missed it release 😅, I'll amend it.

bennymeg, 2024-04-20:
I have checked and this seems not to be the case. I did not check if the "Exclude from board" flag is set.
Would you by any chance set the value to "DNP" or set the "Do not populate" flag on these components?

CodeLima, 2024-04-22:
It looks like they don't officially support this feature. Only if requested via customer support. I did modify the BOM manually and the parser on JLC discards all components who don't have a footprint. 
I did only select "exclude from board" and did not link a footprint. I had a look on your code who seems to take the information for the BOM form the PCB. Since the component never make it to the PCB and will only be available in the schematic, the parts are not available in the PCB data structure. 
Anywas since there is no official support for this form JLC this feature is not really required

bennymeg, 2024-04-25:
This is partially supported now.

You can:
1. Link an "empty footprint" to these parts so they will be synced to the board _*(small but mandatory hack)_.
2. Mark them as DNP so they will not be present in the position file.
3. Do not tick the "Exclude DNP from BOM" checkbox in the plugin.

closed 2024-04-25 by bennymeg